CauldronDevelopmentLLC/CAMotics #356: Failed build on Windows 10

psestan, 2022-01-17:
I hope someone can help me shed some light on the a problem I encountered while trying to build on Windows 10 _64.
While I was previously successful in building the package on Ubuntu 20.04, this time I was following the Windows_Build.md guide and managed to get the following error while running `scons -C CAMotics optimize=1` :

```
Python was not found but can be installed from the Microsoft Store: ms-windows-store://pdp/?productid=9NJ46SX7X90PException: Chakra or V8 support is required, please rebuild C! You may need to set CHAKRA_CORE_HOME or V8_HOME.:
  File "C:\build\CAMotics\SConstruct", line 86:
    'Chakra or V8 support is required, please rebuild C! You may '
```
I would greatly appreciate any help or guidance that could point me towards a solution to this.
- I don't understand the Python error as I do have Python 2.7.12 installed on the machine and have double checked the path variable set in ``env.bat``.
- The V8 exception bugs me as well since the ``scons -C v8-3.14.5 mode=release arch=x64 library=static`` command completed witnout any problems and I have set the V8_HOME variable as described in the Windows_Build.md.
- My  ``env.bat`` looks like this:
```
set PATH=C:\Python27;C:\Python27\Scripts;C:\Program Files (x86)\NSIS;%PATH%
set V8_HOME=C:\build\v8-3.14.5
set V8_LIBPATH=C:\build\v8-3.14.5\obj\release
set CBANG_HOME=C:\build\cbang
set TARGET_ARCH=x86_64
set QTDIR=c:\Qt\Qt5.8.0\5.8\msvc2015_64
```
- C! builds but notes some missing components, i am not sure about the significance of any of those so I am including the configuration part of the ``scons -C cbang disable_local="libevent re2" optimize=1`` output below: 
```
scons: Entering directory `c:\build\cbang'
scons: Reading SConscript files ...
   Compiler: cl (msvc)
   Platform: win32
       Mode: msvc
       Arch: x86_64
       Jobs: 3
Checking for C header file zlib.h... no
Need header zlib.h(cached) error: no result
Checking for C header file bzlib.h... no
Need header bzlib.h(cached) error: no result
Checking for C header file expat.h... no
Need C header expat.h(cached) error: no result
Checking for C library ws2_32... yes
Checking for C library sqlite3... no
Need library sqlite3(cached) error: no result
Checking for C header file yaml.h... no
Need header yaml.h(cached) error: no result
Checking for C library snappy... no
Need library snappy(cached) error: no result
Checking for C header file mysql/mysql.h... no
Checking for C library lz4... no
Checking for C header file event.h... no
Need header event.h(cached) error: no result
Checking for C function epoll_create1()... no
Checking for C header file openssl/ssl.h... no
Checking for C library libcrypto... no
Checking for C library libeay32MT... no
Checking for C library libeay32... no
Need openssl(cached) error: no result
Checking for C++ header file ChakraCore.h... no
Need C++ header ChakraCore.h(cached) error: no result
Checking for C library winmm... yes
Checking for C++ header file v8.h... yes
Checking for C++ header file libplatform/libplatform.h... no
Need C++ header libplatform/libplatform.h(cached) error: no result
Checking for C library ws2_32... yes
Checking for C library winmm... yes
Checking for C library setupapi... yes
Checking for C header file valgrind/valgrind.h... no
Checking for C function memmove()... yes
Checking for C header file strings.h... no
```

Full `scons -C CAMotics optimize=1` output:
```
c:\build>scons -C CAMotics package
scons: Entering directory `c:\build\CAMotics'
scons: Reading SConscript files ...
   Compiler: cl (msvc)
   Platform: win32
       Mode: msvc
       Arch: x86_64
       Jobs: 3
Checking for C header file zlib.h... yes
Checking for C library z... yes
Checking for C header file bzlib.h... yes
Checking for C library bz2... yes
Checking for C header file expat.h... yes
Checking for C library libexpatMT... no
Checking for C library expat... yes
Checking for C library ws2_32... yes
Checking for C library sqlite3... yes
Checking for C header file sqlite3.h... yes
Checking for C function sqlite3_backup_init()... yes
Checking for C header file yaml.h... yes
Checking for C library yaml... yes
Checking for C library snappy... no
Need library snappy(cached) error: no result
Checking for C header file mysql/mysql.h... no
Checking for C library lz4... yes
Checking for C header file event.h... no
Need header event.h(cached) error: no result
Checking for C function epoll_create1()... no
Checking for C++ header file ChakraCore.h... no
Need C++ header ChakraCore.h(cached) error: no result
Checking for C library winmm... yes
Checking for C++ header file v8.h... yes
Checking for C++ header file libplatform/libplatform.h... no
Need C++ header libplatform/libplatform.h(cached) error: no result
Checking for C library ws2_32... yes
Checking for C library winmm... yes
Checking for C library setupapi... yes
Checking for C header file valgrind/valgrind.h... no
Checking for C library cbang-boost... yes
Checking for C library cbang... yes
Checking for C++ header file cbang/Exception.h... yes
Python was not found but can be installed from the Microsoft Store: ms-windows-store://pdp/?productid=9NJ46SX7X90PException: Chakra or V8 support is required, please rebuild C! You may need to set CHAKRA_CORE_HOME or V8_HOME.:
  File "C:\build\CAMotics\SConstruct", line 86:
    'Chakra or V8 support is required, please rebuild C! You may '
```

psestan, 2022-01-20:
I managed to follow the problem and the exception being mentioned as happening on line 86 looks to be raised due to a missing ```libplatform.h``` file which is a part of v8.
The part in the CAMotics build output that appears to ultimately be causing the exception is:
```
Checking for C++ header file libplatform/libplatform.h... no
Need C++ header libplatform/libplatform.h(cached) error: no result
```
I firstly concluded that v8 must've been built successfully due to a lack of errors in its SCons output as well as having the ``` 
Checking for C++ header file v8.h... yes``` line in the CAMotics build output, now howerver with the focus on the missing ```libplatform.h``` i am no longer sure about that.
I am including the output of ```scons -C v8-3.14.5 mode=release arch=x64 library=static``` (v8 build) command below in hopes someone can help me shed some light on this problem.

<details>
  <summary>V8 build output</summary>

```  
c:\build>scons -C v8-3.14.5 mode=release arch=x64 library=static
scons: Entering directory `c:\build\v8-3.14.5'
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
cl /Foobj\release\accessors.obj /c src\accessors.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
accessors.cc
cl /Foobj\release\allocation.obj /c src\allocation.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
allocation.cc
cl /Foobj\release\api.obj /c src\api.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
api.cc
cl /Foobj\release\assembler.obj /c src\assembler.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
assembler.cc
cl /Foobj\release\ast.obj /c src\ast.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
ast.cc
cl /Foobj\release\atomicops_internals_x86_gcc.obj /c src\atomicops_internals_x86_gcc.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
atomicops_internals_x86_gcc.cc
cl /Foobj\release\bignum-dtoa.obj /c src\bignum-dtoa.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
bignum-dtoa.cc
cl /Foobj\release\bignum.obj /c src\bignum.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
bignum.cc
cl /Foobj\release\bootstrapper.obj /c src\bootstrapper.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
bootstrapper.cc
cl /Foobj\release\builtins.obj /c src\builtins.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
builtins.cc
cl /Foobj\release\cached-powers.obj /c src\cached-powers.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
cached-powers.cc
cl /Foobj\release\checks.obj /c src\checks.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
checks.cc
cl /Foobj\release\circular-queue.obj /c src\circular-queue.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
circular-queue.cc
cl /Foobj\release\code-stubs.obj /c src\code-stubs.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
code-stubs.cc
cl /Foobj\release\codegen.obj /c src\codegen.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
codegen.cc
cl /Foobj\release\compilation-cache.obj /c src\compilation-cache.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
compilation-cache.cc
cl /Foobj\release\compiler.obj /c src\compiler.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
compiler.cc
cl /Foobj\release\contexts.obj /c src\contexts.cc /GR- /Gy /EHsc /nologo /W3 /WX /wd4311 /wd4302 /wd4312 /wd4351 /wd4355 /wd4800 /wd4091 /MT /O2 /GL /DWIN32 /DV8_TARGET_ARCH_X64 /DENABLE_DEBUGGER_SUPPORT /Isrc /Zi
contexts.cc
[... cut ...]
lib /MACHINE:X64 /LTCG /NOLOGO /OUT:obj\release\v8preparser.lib obj\release\allocation.obj obj\release\bignum-dtoa.obj obj\release\bignum.obj obj\release\cached-powers.obj obj\release\conversions.obj obj\release\diy-fp.obj obj\release\dtoa.obj obj\release\fast-dtoa.obj obj\release\fixed-dtoa.obj obj\release\preparse-data.obj obj\release\preparser-api.obj obj\release\preparser.obj obj\release\scanner.obj obj\release\strtod.obj obj\release\token.obj obj\release\unicode.obj obj\release\utils.obj
scons: done building targets.
```
</details>

youyizhang, 2022-02-24:
I have this problem too, build on win11.

psestan, 2022-02-24:
Hi @youyizhang, I got in contact with @jcoffland and he explained that with the latest version of the source, for now you wont be able to get a successful build following the [Windows_Build.md](https://github.com/CauldronDevelopmentLLC/CAMotics/blob/master/doc/Windows_Build.md) guide due to some missing features of the V8 3.14.5 that are required for the latest CAMotics.
I did manage to build an older version of the source (this one: [62c6e04](https://github.com/CauldronDevelopmentLLC/CAMotics/commit/62c6e04718e93e4766094f283cb2772834a62e09)) without any problems by following the guide, so if you don't need any of the newer features, that might do it for you too.

youyizhang, 2022-02-25:
I build the version you mentioned that, but still print the same error.

psestan, 2022-02-26:
> I build the version you mentioned that, but still print the same error.

I am not sure, but I don't think that kind of exception should be possible on the older code, are you sure you pulled from the specific commit?
You should clone the repository and then do a checkout followed by the specific commit hash, like this:
```git checkout 62c6e04718e93e4766094f283cb2772834a62e09```
Then you can try building it again.

youyizhang, 2022-03-01:
I tried several ways, but all of failed.
1. failed of git checkout
[image]
2. And then I try to use git reset --hard, finally print error of 'Chakra or V8 support is required, please rebuild C! You may '.
[image]
3. I tried to directly downlod the CAMotics-62c6e04718e93e4766094f283cb2772834a62e09.zip, CAMotics-1.1.0-debug.zip and CAMotics-1.1.0-release.zip, but all failed.
I build successfully in ubuntu18.04, but I never code C++ in ubuntu.
Do you have any experience about debugging code in ubuntu? I am planning to add some function  in CAMotics.

luzpaz, 2022-07-10:
Ticket should be labeled installer and windows related

youyizhang, 2022-07-11:
I have successfully build Camotios in Window.
The libraries version are:
CAMotics-1.2.0-debug
C! (1.3.1)