CauldronDevelopmentLLC/CAMotics #359: ReactOS And Latest ubuntu 22.04 fedora 35 and later opensuse leap 15.3 support?

larsmartin4, 2022-03-19:
What is reason that ReactOS and latest ubuntu fedora 35 and opensuse leap 15.3 did not support this app? its hard to install but its did not works.

Only linux mint and windows 10 works maybe windows 7? And older ubuntu 21.04 or older.

jcoffland, 2022-03-19:
I don't know.  I'm not involved in maintaining those packages.  You'd have to ask the package maintainers.

closed 2022-03-19 by jcoffland

larsmartin4, 2022-03-19:
who is package maintainers?

jcoffland, 2022-03-19:
Possibly @srakitnican.

larsmartin4, 2022-03-20:
Also look at: https://github.com/CauldronDevelopmentLLC/camotics-web/issues/2 its need more information what is support or not.

RWL-69, 2024-07-03:
Now 2024 and there is no rpm package.  No maintainers or it is just too hard.  

No instructions to compile for none Debian based systems.

larsmartin4, 2024-07-03:
Its possible to use old Ubuntu og mint debian.

But now is it ubuntu 24.04 or fedora 41 soon posible to test fedora 42 alpha.

Its can be used tar or copy old need files from old distribution but not sure if most system have comply layer can be used old system.

Its possible to try to get exe to fix on wine?
Works with windows.

So what its its you idea?

VMware players use .bundle for Linux.

You can find rpm package camotics-1.2.0.x86_64.rpm but its for old distribution that is out of date today.need to fix library to new system.

Lars Martin Hambro