CauldronDevelopmentLLC/CAMotics #365: CAMotics and Ubuntu 22.04lts

StevenTyrer, 2022-05-03:
Have been using CAMotics on Ubuntu 21.10 with no problems but on upgrading to Ubuntu 22.04tls it will not run, I have tried removing and reinstalling CAMotics no with improvement I have also installed the  addition programs as recommended on in the installation guide.
and it will still not run.

Steven

jcoffland, 2022-05-05:
Please try running CAMotics from a terminal and post any errors here.

    camotics

Also try running this command:

    ldd /usr/bin/camotics

And post the output here.

StevenTyrer, 2022-05-05:
As requested

steven@steven-Laptop:~/.config/Cauldron Development LLC$ camotics
camotics: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory
===========================================================================================
steven@steven-Laptop:~/.config/Cauldron Development LLC$ ldd /usr/bin/camotics
	linux-vdso.so.1 (0x00007ffcc47a7000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f4a57d0e000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007f4a57c87000)
	libv8.so.3.14.5 => /usr/lib/libv8.so.3.14.5 (0x00007f4a57600000)
	libssl.so.1.1 => not found
	libcrypto.so.1.1 => not found
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f4a57c80000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f4a57c7b000)
	libQt5OpenGL.so.5 => /lib/x86_64-linux-gnu/libQt5OpenGL.so.5 (0x00007f4a57c1c000)
	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f4a56f3f000)
	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f4a56861000)
	libQt5WebSockets.so.5 => /lib/x86_64-linux-gnu/libQt5WebSockets.so.5 (0x00007f4a56836000)
	libQt5Network.so.5 => /lib/x86_64-linux-gnu/libQt5Network.so.5 (0x00007f4a56685000)
	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f4a56126000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f4a5603f000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f4a5601f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4a55df7000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f4a57f58000)
	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f4a55d3f000)
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f4a55d0b000)
	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f4a55cd0000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f4a55cb4000)
	libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f4a55be5000)
	libmd4c.so.0 => /lib/x86_64-linux-gnu/libmd4c.so.0 (0x00007f4a55bd3000)
	libgssapi_krb5.so.2 => /lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x00007f4a55b7f000)
	libdouble-conversion.so.3 => /lib/x86_64-linux-gnu/libdouble-conversion.so.3 (0x00007f4a55b6a000)
	libicui18n.so.70 => /lib/x86_64-linux-gnu/libicui18n.so.70 (0x00007f4a5583b000)
	libicuuc.so.70 => /lib/x86_64-linux-gnu/libicuuc.so.70 (0x00007f4a55640000)
	libpcre2-16.so.0 => /lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x00007f4a555b7000)
	libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x00007f4a554e8000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f4a553ae000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f4a5526e000)
	libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f4a551a6000)
	libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f4a5517f000)
	libkrb5.so.3 => /lib/x86_64-linux-gnu/libkrb5.so.3 (0x00007f4a550b4000)
	libk5crypto.so.3 => /lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007f4a55085000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007f4a57c0c000)
	libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007f4a55077000)
	libicudata.so.70 => /lib/x86_64-linux-gnu/libicudata.so.70 (0x00007f4a53459000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f4a533e3000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f4a533b9000)
	libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x00007f4a533ab000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007f4a533a4000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f4a53390000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007f4a5338a000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f4a53380000)
	libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x00007f4a5335d000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f4a53345000)
	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007f4a53338000)

jcoffland, 2022-05-06:
The problem is that it cannot find the OpenSSL version 1.1 libraries.  CAMotics needs this to be installed.  It looks like Ubuntu moved to OpenSSL version 3.  You could try this:

```
curl "http://security.ubuntu.com/ubuntu/pool/main/o/openssl/{libssl1.1_1.1.1f-1ubuntu2_amd64.deb}" -o "#1"
sudo dpkg -i --force-all libssl1.1_1.1.1f-1ubuntu2_amd64.deb
```

StevenTyrer, 2022-05-06:
Thanks Joseph, that worked.
Steven

closed 2022-05-06 by jcoffland