CauldronDevelopmentLLC/CAMotics #368: Feature Request - perpendicular slice viewer

inspectionsbybob, 2022-05-25:
I am looking for a platform that either has a feature or would be a good place to start for "Perpendicular slice viewer"

Almost every slicer has a Gcode preview tool.  The issue I have is that it only allows viewing of the exterior -OR- a single horizontal slice in the plane it is printed in.

I was working on a messy model with lots of depressions, overhangs, and unusual shapes needing supports and infill.  Looking at the horizontal slices it was not clear if infill was going to adequately support some features.  What I wanted was a vertical CUT or cross section of the model so I could see all the structures. Some "xray" viewers come close, but they don't support vertical browsing

I know there are VERY High-end stress simulation softwares that can provide some feedback about models but that is overkill for most hobbist and general users who just want to see what the infill is doing, especially when you start adding infill modifiers.

love to work with a group if they think it has merit developing the spec and then working on the development.

jcoffland, 2022-05-25:
This shouldn't be too difficult.  The UI parts are probably the most difficult.

inspectionsbybob, 2022-05-25:
Actually I found one... not sure if it is open source....

Was many many many pages down in the search...

https://craftbot.com/craftware

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016