CauldronDevelopmentLLC/CAMotics #369: Gcc11 failed to build

jack9603301, 2022-05-26:
When I finally tried to build this project, scons reported the following error:

```
('build/resources.cpp', 'src/resources.data/data2.cpp', 'src/resources.data/data3.cpp')
scons: done reading SConscript files.
scons: Building targets ...
build_function(["build/build_info.cpp"], [])
g++ -o build/camotics/contour/TriangleMesh.o -c -faligned-new -std=c++17 -fsigned-char -Wno-deprecated-declarations -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DHAVE_CBANG -DUSING_CBANG -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_NETWORK_LIB -DQT_WIDGETS_LIB -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -I/home/jack/SourceCode/CAMotics/cbang/src -I/home/jack/SourceCode/CAMotics/cbang/include -I/home/jack/SourceCode/CAMotics/cbang/src/boost -Isrc -I/usr/include/python3.10 -I/usr/include/qt5/QtCore -I/usr/include/qt5 -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtOpenGL -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtWebSockets -Ibuild -Isrc src/camotics/contour/TriangleMesh.cpp
g++ -o build/camotics/contour/VertexSlice.o -c -faligned-new -std=c++17 -fsigned-char -Wno-deprecated-declarations -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DHAVE_CBANG -DUSING_CBANG -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_NETWORK_LIB -DQT_WIDGETS_LIB -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -I/home/jack/SourceCode/CAMotics/cbang/src -I/home/jack/SourceCode/CAMotics/cbang/include -I/home/jack/SourceCode/CAMotics/cbang/src/boost -Isrc -I/usr/include/python3.10 -I/usr/include/qt5/QtCore -I/usr/include/qt5 -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtOpenGL -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtWebSockets -Ibuild -Isrc src/camotics/contour/VertexSlice.cpp
g++ -o build/camotics/machine/MachineModel.o -c -faligned-new -std=c++17 -fsigned-char -Wno-deprecated-declarations -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DHAVE_CBANG -DUSING_CBANG -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_NETWORK_LIB -DQT_WIDGETS_LIB -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -I/home/jack/SourceCode/CAMotics/cbang/src -I/home/jack/SourceCode/CAMotics/cbang/include -I/home/jack/SourceCode/CAMotics/cbang/src/boost -Isrc -I/usr/include/python3.10 -I/usr/include/qt5/QtCore -I/usr/include/qt5 -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtOpenGL -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtWebSockets -Ibuild -Isrc src/camotics/machine/MachineModel.cpp
g++ -o build/camotics/machine/MachinePart.o -c -faligned-new -std=c++17 -fsigned-char -Wno-deprecated-declarations -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DHAVE_CBANG -DUSING_CBANG -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_NETWORK_LIB -DQT_WIDGETS_LIB -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -I/home/jack/SourceCode/CAMotics/cbang/src -I/home/jack/SourceCode/CAMotics/cbang/include -I/home/jack/SourceCode/CAMotics/cbang/src/boost -Isrc -I/usr/include/python3.10 -I/usr/include/qt5/QtCore -I/usr/include/qt5 -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtOpenGL -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtWebSockets -Ibuild -Isrc src/camotics/machine/MachinePart.cpp
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/include/g++-v11/set:60,
                 from /home/jack/SourceCode/CAMotics/cbang/src/cbang/json/NullSink.h:39,
                 from /home/jack/SourceCode/CAMotics/cbang/src/cbang/json/Writer.h:35,
                 from /home/jack/SourceCode/CAMotics/cbang/src/cbang/json/Value.h:37,
                 from /home/jack/SourceCode/CAMotics/cbang/src/cbang/json/List.h:35,
                 from /home/jack/SourceCode/CAMotics/cbang/src/cbang/geom/Vector.h:39,
                 from src/camotics/contour/TriangleMesh.h:23,
                 from src/camotics/contour/TriangleMesh.cpp:21:
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/include/g++-v11/bits/stl_tree.h: In instantiation of ‘static const _Key& std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_S_key(std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_Const_Link_type) [with _Key = CAMotics::TriangleMesh::Vertex*; _Val = CAMotics::TriangleMesh::Vertex*; _KeyOfValue = std::_Identity<CAMotics::TriangleMesh::Vertex*>; _Compare = CAMotics::TriangleMesh::VertexSort; _Alloc = std::allocator<CAMotics::TriangleMesh::Vertex*>; std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_Const_Link_type = const std::_Rb_tree_node<CAMotics::TriangleMesh::Vertex*>*]’:
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/include/g++-v11/bits/stl_tree.h:2071:47:   required from ‘std::pair<std::_Rb_tree_node_base*, std::_Rb_tree_node_base*> std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_get_insert_unique_pos(const key_type&) [with _Key = CAMotics::TriangleMesh::Vertex*; _Val = CAMotics::TriangleMesh::Vertex*; _KeyOfValue = std::_Identity<CAMotics::TriangleMesh::Vertex*>; _Compare = CAMotics::TriangleMesh::VertexSort; _Alloc = std::allocator<CAMotics::TriangleMesh::Vertex*>; std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::key_type = CAMotics::TriangleMesh::Vertex*]’
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/include/g++-v11/bits/stl_tree.h:2124:4:   required from ‘std::pair<std::_Rb_tree_iterator<_Val>, bool> std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_insert_unique(_Arg&&) [with _Arg = CAMotics::TriangleMesh::Vertex* const&; _Key = CAMotics::TriangleMesh::Vertex*; _Val = CAMotics::TriangleMesh::Vertex*; _KeyOfValue = std::_Identity<CAMotics::TriangleMesh::Vertex*>; _Compare = CAMotics::TriangleMesh::VertexSort; _Alloc = std::allocator<CAMotics::TriangleMesh::Vertex*>]’
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/include/g++-v11/bits/stl_set.h:512:25:   required from ‘std::pair<typename std::_Rb_tree<_Key, _Key, std::_Identity<_Tp>, _Compare, typename __gnu_cxx::__alloc_traits<_Alloc>::rebind<_Key>::other>::const_iterator, bool> std::set<_Key, _Compare, _Alloc>::insert(const value_type&) [with _Key = CAMotics::TriangleMesh::Vertex*; _Compare = CAMotics::TriangleMesh::VertexSort; _Alloc = std::allocator<CAMotics::TriangleMesh::Vertex*>; typename std::_Rb_tree<_Key, _Key, std::_Identity<_Tp>, _Compare, typename __gnu_cxx::__alloc_traits<_Alloc>::rebind<_Key>::other>::const_iterator = std::_Rb_tree<CAMotics::TriangleMesh::Vertex*, CAMotics::TriangleMesh::Vertex*, std::_Identity<CAMotics::TriangleMesh::Vertex*>, CAMotics::TriangleMesh::VertexSort, std::allocator<CAMotics::TriangleMesh::Vertex*> >::const_iterator; typename __gnu_cxx::__alloc_traits<_Alloc>::rebind<_Key>::other = std::allocator<CAMotics::TriangleMesh::Vertex*>; typename __gnu_cxx::__alloc_traits<_Alloc>::rebind<_Key> = __gnu_cxx::__alloc_traits<std::allocator<CAMotics::TriangleMesh::Vertex*>, CAMotics::TriangleMesh::Vertex*>::rebind<CAMotics::TriangleMesh::Vertex*>; typename _Alloc::value_type = CAMotics::TriangleMesh::Vertex*; std::set<_Key, _Compare, _Alloc>::value_type = CAMotics::TriangleMesh::Vertex*]’
src/camotics/contour/TriangleMesh.cpp:109:50:   required from here
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/include/g++-v11/bits/stl_tree.h:770:15: 错误：static assertion failed: comparison object must be invocable as const
  770 |               is_invocable_v<const _Compare&, const _Key&, const _Key&>,
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/include/g++-v11/bits/stl_tree.h:770:15: 附注：‘std::is_invocable_v<const CAMotics::TriangleMesh::VertexSort&, CAMotics::TriangleMesh::Vertex* const&, CAMotics::TriangleMesh::Vertex* const&>’ evaluates to false
scons: *** [build/camotics/contour/TriangleMesh.o] Error 1

```

Does anyone know how to solve it? I use Gentoo source code Linux distribution!

jack9603301, 2022-05-26:
I don't know why I use C + + 17. I can't compile with C + + 14 and can't detect V8

jack9603301, 2022-05-26:
@jcoffland If I build with c++11, an error like the following occurs and I can't even get it to detect properly

```
Checking for C header file pthread.h... (cached) yes
Checking for C library pthread... yes
Checking for C header file valgrind/valgrind.h... no
Checking for C library cbang-boost... yes
Checking for C library cbang... yes
Checking for C++ header file cbang/Exception.h... yes
Checking for C header file Python.h... (cached) yes
Checking for C function Py_Initialize()... yes
Exception: Chakra or V8 support is required, please rebuild C! You may need to set CHAKRA_CORE_HOME or V8_HOME.:
  File "/home/jack/SourceCode/CAMotics/CAMotics/SConstruct", line 85:
    raise Exception(

```

jcoffland, 2022-05-26:
Yes, you have to either rebuild cbang with ``libv8`` support or disable tpl like this:

    scons with_tpl=0

jack9603301, 2022-05-26:
I have used the official fork version of v8 to compile and install, and compile cbang, still useless

jcoffland, 2022-05-26:
Are you sure C! compiled with v8?  During the config phase you should see:

```
Checking for C++ header file v8.h... (cached) yes
Checking for C++ header file libplatform/libplatform.h... (cached) yes
Checking for C library v8... (cached) yes
Checking for C library v8_libplatform... (cached) yes
```

Possibly with out ``(cached)``.  You should also find ``#define HAVE_V8`` in ``cbang/include/cbang/config.h``.  If not, then cbang did not compile with v8 support.

On Debian I install the ``libnode-dev`` package to get ``libv8``.  I'm not sure that https://github.com/CauldronDevelopmentLLC/v8-3.14.5 works with cbang any more.  It's a rather old version.

jack9603301, 2022-05-26:
@jcoffland 
```
sr/include/qt5/QtOpenGL -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtWebSockets -Ibuild -Isrc src/planner.cpp
ranlib build/libCAMotics.a
g++ -o build/python.os -c -faligned-new -std=c++11 -fsigned-char -Wno-deprecated-declarations -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -fPIC -DNDEBUG -D_REENTRANT -DHAVE_CBANG -DUSING_CBANG -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_NETWORK_LIB -DQT_WIDGETS_LIB -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -I/home/jack/SourceCode/CAMotics/cbang/src -I/home/jack/SourceCode/CAMotics/cbang/include -I/home/jack/SourceCode/CAMotics/cbang/src/boost -Isrc -I/usr/include/python3.10 -I/usr/include/qt5/QtCore -I/usr/include/qt5 -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtOpenGL -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtWebSockets -Ibuild -Isrc src/python.cpp
ar rc build/libCAMoticsPy.a build/python/PyJSON.o build/python/PyJSONSink.o build/python/PyLogger.o build/python/PyNameResolver.o build/python/PyPlanner.o build/python/PySimulation.o build/python/PyTask.o
ranlib build/libCAMoticsPy.a
g++ -o gcodetool -Wl,--as-needed -Wl,-s -Wl,-x -Wl,--gc-sections -no-pie -pthread -Wl,-rpath=/usr/lib build/gcodetool.o -L/home/jack/SourceCode/CAMotics/cbang/lib -L. -Wl,--start-group build/libCAMoticsPy.a build/libCAMotics.a build/libCAMoticsGUI.a build/libDXF.a build/libSTL.a build/libGCode.a -lstdc++ -lutil -lcbang -lcbang-boost -lssl -lcrypto -llz4 -lmysqlclient -lexpat -lbz2 -lz -lpthread -ldl -lm -lpython3.10 build/dxflib/libdxflib.a -Wl,--end-group
g++ -o build/camotics.so -Wl,--as-needed -Wl,-s -Wl,-x -Wl,--gc-sections -no-pie -pthread -shared -Wl,-rpath=/usr/lib build/python.os -L/home/jack/SourceCode/CAMotics/cbang/lib -L. -Wl,--start-group build/libCAMoticsPy.a build/libCAMotics.a build/libCAMoticsGUI.a build/libDXF.a build/libSTL.a build/libGCode.a -lstdc++ -lutil -lcbang -lcbang-boost -lssl -lcrypto -llz4 -lmysqlclient -lexpat -lbz2 -lz -lpthread -ldl -lm -lpython3.10 build/dxflib/libdxflib.a -Wl,--end-group
g++ -o planner -Wl,--as-needed -Wl,-s -Wl,-x -Wl,--gc-sections -no-pie -pthread -Wl,-rpath=/usr/lib build/planner.o -L/home/jack/SourceCode/CAMotics/cbang/lib -L. -Wl,--start-group build/libCAMoticsPy.a build/libCAMotics.a build/libCAMoticsGUI.a build/libDXF.a build/libSTL.a build/libGCode.a -lstdc++ -lutil -lcbang -lcbang-boost -lssl -lcrypto -llz4 -lmysqlclient -lexpat -lbz2 -lz -lpthread -ldl -lm -lpython3.10 build/dxflib/libdxflib.a -Wl,--end-group
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: build/libGCode.a(TPLRunner.o): in function `GCode::TPLRunner::run()':
TPLRunner.cpp:(.text._ZN5GCode9TPLRunner3runEv+0xbb): undefined reference to `tplang::TPLContext::TPLContext(cb::SmartPointer<std::ostream, cb::DeallocNew<std::ostream>, cb::RefCounterImpl<std::ostream, cb::DeallocNew<std::ostream> > > const&, GCode::MachineInterface&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: 错误：ld 返回 1
```

jack9603301, 2022-05-26:
> Are you sure C! compiled with v8? During the config phase you should see:
>
> ```
> Checking for C++ header file v8.h... (cached) yes
> Checking for C++ header file libplatform/libplatform.h... (cached) [...]

OK, can you confirm your version number of V8? I have to compile and install because I am Gentoo

jcoffland, 2022-05-26:
I've just pushed a change that fixes the ``with_tpl=0`` build mode if you want to try that again.

jack9603301, 2022-05-27:
@jcoffland I'm currently trying to build a cbang under v8 7.8.285, since the two components are linked, I don't think it's necessary to open a separate issue, I'm getting the following error:

```
g++ -o build/cbang/js/Sink.o -c -faligned-new -std=c++11 -fsigned-char -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DUSING_CBANG -Iinclude -Isrc -Isrc/boost src/cbang/js/Sink.cpp
In file included from src/cbang/js/v8/ValueRef.h:35,
                 from src/cbang/js/v8/JSImpl.h:35,
                 from src/cbang/js/Javascript.cpp:40:
src/cbang/js/v8/Value.h: 在成员函数‘virtual bool cb::gv8::Value::toBoolean() const’中:
src/cbang/js/v8/Value.h:90:42: 错误：cannot convert ‘v8::Local<v8::Context>’ to ‘v8::Isolate*’
   90 |         return value->BooleanValue(getCtx()).FromJust();
      |                                    ~~~~~~^~
      |                                          |
      |                                          v8::Local<v8::Context>
In file included from src/cbang/js/v8/V8.h:45,
                 from src/cbang/js/v8/Value.h:35,
                 from src/cbang/js/v8/ValueRef.h:35,
                 from src/cbang/js/v8/JSImpl.h:35,
                 from src/cbang/js/Javascript.cpp:40:
/usr/include/v8.h:2639:30: 附注：  初始化‘bool v8::Value::BooleanValue(v8::Isolate*) const’的实参 1
 2639 |   bool BooleanValue(Isolate* isolate) const;
      |                     ~~~~~~~~~^~~~~~~
g++ -o build/cbang/js/StdModule.o -c -faligned-new -std=c++11 -fsigned-char -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DUSING_CBANG -Iinclude -Isrc -Isrc/boost src/cbang/js/StdModule.cpp
scons: *** [build/cbang/js/Javascript.o] Error 1
scons: building terminated because of errors.

```

jcoffland, 2022-05-27:
I pushed a commit that should fix the above c! build problem.  The link errors maybe related.

jack9603301, 2022-05-27:
@jcoffland Now there are still link errors. Let me check whether there is a problem with the system. It is strange that I use Qt5 on Gentoo and will report such link errors.

jack9603301, 2022-05-28:
@jcoffland 

I still have this error, I have Qt and icu installed, there shouldn't be a problem, I'm running KDE

```
scons: *** [camsim] Error 1
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /home/jack/SourceCode/CAMotics/cbang/lib/libcbang.a(JSImpl.o): in function `cb::gv8::JSImpl::init(int*, char**)':
JSImpl.cpp:(.text._ZN2cb3gv86JSImpl4initEPiPPc+0x8c): undefined reference to `v8::platform::NewDefaultPlatform(int, v8::platform::IdleTaskSupport, v8::platform::InProcessStackDumping, std::unique_ptr<v8::TracingController, std::default_delete<v8::TracingController> >)'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucnv_getStandardName'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucnv_setSubstChars'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucal_getDSTSavings'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucnv_countAvailable'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucal_getDefaultTimeZone'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucnv_fromUnicode'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucal_close'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `u_errorName'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucal_openCountryTimeZones'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucol_setAttribute'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucnv_countAliases'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucnv_getDefaultName'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucnv_toUnicode'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucal_getTimeZoneDisplayName'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucal_openTimeZones'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucol_strcoll'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `u_strToUpper'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucnv_compareNames'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucnv_getAvailableName'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucnv_getAlias'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucal_inDaylightTime'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucol_getSortKey'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `uenum_close'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucal_open'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucal_get'
[... cut ...]
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucnv_getMaxCharSize'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `u_strToLower'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../lib64/libQt5Core.so: undefined reference to `ucnv_open'
collect2: 错误：ld 返回 1

```

jack9603301, 2022-05-28:
@jcoffland It seems that I have found the root cause of the link error. During the link, the ICU version used by V8 is different from the ICU version installed in the system, which leads me to overwrite the ICU version of the system by mistake (when I use CP). The possible solution may be to adopt static link, or use the system ICU (but I can't find this method), or disable the ICU of V8

jack9603301, 2022-05-28:
Last solution: disable v8

hpmachining, 2022-06-29:
@jack9603301 Here is a [link](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=v8-r) to an Arch Linux PKGBUILD file that might help you figure out how to build v8 with the system icu. You need to use c++17 for cbang and camotics to recognize the latest v8 (10.5.84). There was 1 minor change I needed to make to camotics to get it to build with c++17. See this [pull request](https://github.com/CauldronDevelopmentLLC/CAMotics/pull/372)

closed 2023-08-10 by jcoffland