CauldronDevelopmentLLC/CAMotics #377: SConstruct question

sergey-239, 2022-09-18:
Dear Joseph,
there is the following code in SConstruct at lines 317 through 331:
```
    # Examples
    examples = []
    cmd = 'git ls-files examples/'
    p = subprocess.Popen(cmd, shell = True, stdout = subprocess.PIPE)
    examples = p.communicate()[0]
    if isinstance(examples, bytes): examples = examples.decode()
    examples = list(map(lambda x: [x, x], examples.split()))

    # Machines
    machines = []
    cmd = 'git ls-files machines/'
    p = subprocess.Popen(cmd, shell = True, stdout = subprocess.PIPE)
    machines = p.communicate()[0]
    if isinstance(machines, bytes): machines = machines.decode()
    machines = list(map(lambda x: [x, x], machines.split()))
```
I am not an expert in Python, so could you, please, clarify, what was the intent? Could this build dependency on `git` to be removed? 
Thanks in advance

sergey-239, 2022-09-18:
Sorry, Jozeph,
the annoying warnings about missing `git` where not from this source.

sergey-239, 2022-09-18:
Sorry, Jozeph,
the annoying warnings about missing git where not from this source.

closed 2022-09-18 by sergey-239

jcoffland, 2022-09-18:
To clarify those lines are getting the list of example and machine files which are checked in to git.  This avoids accidentally packaging any temporary files or works in progress.