CauldronDevelopmentLLC/CAMotics #385: Python API simulation start() fails

nikcleju, 2023-03-05:
I'm trying to run one of the examples via the Python API. I have something like this:

```
import sys
from time import sleep
sys.path.append('/usr/lib/python3/dist-packages')

import camotics

sim = camotics.Simulation()

sim.open('/usr/share/doc/camotics/examples/wave/wave.camotics')

with open("/usr/share/doc/camotics/examples/wave/wave.gcode") as f:
    gcode = '\n'.join(f.readlines())

sim.compute_path(gcode=gcode)
while sim.is_running():
    print("Sleep 1")
    sleep(1)

print("Start sim...")
sim.start()
```

When calling `sim.start()`, it terminates with the following output:
```
Start sim...
pure virtual method called
terminate called without an active exception
```

Do you have an yidea what could be wrong?

CAMotics is latest from mainline, compiled from source. I tried with a latest CBang, and also with an older 1.6.1 version, same problem.

nikcleju, 2023-03-05:
I realized I should probably call `wait()` after each call.

```
sim.open('/usr/share/doc/camotics/examples/wave/wave.camotics')
sim.wait()

with open("/usr/share/doc/camotics/examples/wave/wave.gcode") as f:
    gcode = '\n'.join(f.readlines())

sim.compute_path(gcode=gcode)
sim.wait()

print("Start sim...")
sim.start()
sim.wait()
print("Done)
```

In this case `sim.start()` dies with a Segmentation Fault.

nikcleju, 2023-03-05:
I managed to get it running by replacing the `sim.wait()` after `sim.start()` with:

```python
sim.start(threads=4, time=100000, reduce=True)
#sim.wait()
while sim.is_running():
    sleep(0.01)
```

With `sim.wait()` it would sometimes work, and sometimes not. I still don't understand why is this happening.

jcoffland, 2023-03-09:
The problem is that there is a mistake in ``wait()``.  It calls ``PyTask::join()`` which interrupts the task then waits for it to exit.  It will be fixed in the next release. What you're doing with the while loop is fine.