CauldronDevelopmentLLC/CAMotics #386: Problems on Windows with umlauts ä ö ü in path name to g code file

ModuloFS, 2023-04-01:
Hello,

if there is a umlaut like ä, ü, or ö in any position of the path to the stored g-code file. The simulation of the path is not loaded / working. A Error message is not shown, because of that it takes some time to find the failure for a new user. 

Anybody else has the same problem?

gadelan, 2023-11-18:
Yes. It also fails when using other extended latin-1 characters like á é ç in the .camotics project file path. Also Windows.

maciek001, 2026-10-26:
Hi, there is also an issue with Polish characters such as 'ą', 'ę', 'ż', etc. If such characters appear in a file path, CAMotics may create its own set of folders, replacing, for example, the character 'ą' with 'Ä…'. Ver 1.2.0 64bit for Wind 11. 

It appears that an error occurs as early as the stage of opening a project from a directory whose path contains these characters. After adding an .nc file and saving the project, CAMotics creates a new set of folders—as if it failed to recognize the correct path and the project file located within it.

maciek001, 2026-10-30:
@jcoffland I’ve done a quick review of the code. Could you make a one adjustment and check whether at least some of the errors have been fixed? I have no way to build the code.

src/camotics/qt/FileTabManager.cpp line 198: QString path = file.getPath().c_str() -> QString path = QString::fromUtf8(file.getPath().c_str());