CauldronDevelopmentLLC/CAMotics #387: tplang.exe crash on a line with "tool(1);" loading a tpl file

kvrban, 2023-04-14:
On windows 10 tplang.exe crash on a line with "tool(1);" in a tpl file. Without "tool" it works fine.

```
PS C:\Program Files (x86)\CAMotics\examples\camotics> ../../tplang.exe 'C:\Users\kris\kris2.tpl'
G21
(File: C:\Users\kris\kris2.tpl)
F400
M6 T1
ERROR:Exception: SmartPointer: Can't dereference a NULL pointer!
ERROR:       At: C:\Users\kris\kris2.tpl:2:0
```

I use the sample code from https://tplang.org/#examples

```
feed(400);     // Set the feed rate to 400 millimeters per minute
tool(1);       // Select tool 1

rapid({z: 5}); // Move to a safe height of 5mm
rapid({x: 1, y: 1});  // Go to start position
speed(2000);   // Spin at 2000 RPM in the clockwise direction

cut({z: -3});  // Cut down to depth
cut({x: 11});  // Cut to second corner
cut({y: 11});  // Cut to third corner
cut({x: 1});   // Cut to forth corner
cut({y: 1});   // Cut back to begining

rapid({z: 5}); // Move back to safe position
speed(0);      // Stop spinning
```

kvrban, 2023-04-14:
Ok, with tool(); it is working. So a tool no.1 must be defined i guess to not crash. Unpleasant that it crashes, but nothing bad.

closed 2023-04-14 by kvrban

jcoffland, 2023-04-15:
Yes, the exception is unfortunate.  ``tplang`` expects you to pass a JSON file via the ``--sim-json`` option that defines simulation parameters such as the tool table.  You can export a simulation JSON file for a project from the CAMotics GUI.  This will also show you the format.

kvrban, 2023-04-15:
> pass a JSON file via the --sim-json option
Perfect. This is exactly the information I was missing. Thank you