CauldronDevelopmentLLC/CAMotics #389: src/glibc.c and compat_glibc wrappers in v1.3.0

thierry-FreeBSD, 2023-04-18:
Hello,
What is the aim of the file src/glibc.c?
It defines wrappers for __GNUC__ but not for __clang__, and thus, trying to build CAMotics on FreeBSD with the default compiler (clang) fails with theses errors;
```
ld: error: undefined symbol: __wrap_memcpy
ld: error: undefined symbol: __wrap_pow
ld: error: undefined symbol: __wrap_exp
etc.
```
and I guess that the same errors occurs on other systems like Mac or *BSD.

jcoffland, 2023-04-20:
Wrapping a few glibc functions allows us to target older systems while still using a fairly new compiler.  It's not at all necessary when you are building on the target system.  We don't support *BSD directly.  It does build on macOS.  The wrapping only occurs when the scons platform is ``posix``.

I've added a build option which you can use to disable glibc wrapping.

    scons wrap_glibc=0

closed 2023-04-20 by jcoffland

thierry-FreeBSD, 2023-04-20:
Thanks for the quick fix! It builds and runs fine.

Just a minor remark: by default, on my workstation the locale is usually set to fr_FR.UTF8, which has a comma in place of the decimal point, and it produces an error in the examples (invalid JSON number): I have to switch to a different locale, e.g. en_US.UTF8.

jcoffland, 2023-04-20:
Interesting.  The software sets ``LC_NUMERIC=C`` which is supposed to solve that problem.  Which program is it that has this problem?  Is it ``camotics`` or one of the other commands?

thierry-FreeBSD, 2023-04-20:
Yes, this is camotics.