CauldronDevelopmentLLC/CAMotics #395: save the resolution

tcurdt, 2023-06-04:
First off:
hanks for CAMotics. It such a useful tool for me atm.

One thing that would be great to change:

Whenever I start CAMotics the resolution jumps backs to "Medium".
I always have to go into the settings and change it.

I would expect that when "Save" it would be stored across restarts.

[image]

It would be great if the "Resolution" would actually be saved.

tcurdt, 2023-08-23:
It seems the machine type is saved - but the resolution is not.
I was going through the code but didn't understand the cause for that difference.
Could you point into the right direction here?

jcoffland, 2023-08-24:
The resolution is saved with the project.  If you save and reopen a project the resolution should stay the same.  I idea is that the appropriate resolution is project dependent because projects exist at different scales.  However, the settings ``Low``, ``Medium``, etc. are calculated relative to the projects maximum dimension.

tcurdt, 2023-08-24:
Maybe to explain my workflow and there is a better way to go about this:

I write some gcode and open it up via command line CAMotics. On a macOS thats:

    open -a CAMotics --args `pwd`/file.nc
    
On every open I get greeted with 

[image]

I guess if the gcode would reload when I change `file.nc` using a project could become viable.
Then storying it in the project would make sense. And I would have to always click on the dialog.

Is there a better way for my workflow that could solve this the way you thought about CAMotics?

jcoffland, 2023-08-24:
If you open a file in CAMotics and simulate it.  You can change that file on disk and then just hit the reload button.  It will resimulate with the updated file.  One issue though, if you edit the file outside of the CAMotics editor, the changes will not be reflected in the CAMotics GCode editor.

If you're just manually editing the GCode, i.e. not generating it from some other program, you could just do you editing with in CAMotics.  Double click on the file to open the GCode editor.

The third option is to write a project file with the resolution set to what you want.

tcurdt, 2023-08-24:
> One issue though, if you edit the file outside of the CAMotics editor, the changes will not be reflected in the CAMotics GCode editor.

That is the problem. I am generating the gcode file and would need a way to reload it from disk - somehow. Ideally automatic. Then I could just keep the project around with the correct settings.

jcoffland, 2023-08-24:
But the simulation will reflect the changes.

You can also create a project file.  Save it.  Then update your GCode file and reopen CAMotics using the project file.

tcurdt, 2023-08-24:
> reopen CAMotics using the project file

That would be pretty nice. But it seems to just ignore the request to re-open and keeps the old file. I would be nice if that would trigger a reload. Right now only after closing and then re-open or pressing "reload" the new file is being used.

At least it sounds like this could be a pretty simple change.

Nevertheless. I think it would be nice if the defaults where somehow configurable.
That would also allows to get rid of the "create project dialog.

Thanks for all your input BTW.