CauldronDevelopmentLLC/CAMotics #401: FEATURE: Suggested Patch to Add Picking To TPL Files

50J0uRNeR, 2023-08-10:
Greetings,

I've been working on a way to add picking support to TPL files in CAMotics. Unfortunately I could not find a way to sync the gcode stream with the path moves. It would've been nice to see the gcode stream and the picked line in a text tab or console. However I found a really simple way to get picking and the move dock working with TPL. Since this is only a few lines of code I thought it would be easier to post a patch here instead of doing a full pull request. Please let me know if this is inappropriate and I won't do it again ;-)

These patches should work with commit a515a8b:

```
--- /home/sojourner/Sources/CAMotics/CAMotics/src/gcode/machine/MoveSink.h
+++ /home/sojourner/Sources/CAMotics/CAMotics_PATCHED/src/gcode/machine/MoveSink.h
@@ -34,6 +34,7 @@
 
     bool probePending = false;
     double time = 0;
+    unsigned moveCount = 0;
     cb::SmartPointer<std::string> lastFile;
 
   public:
```

```
--- /home/sojourner/Sources/CAMotics/CAMotics/src/gcode/machine/MoveSink.cpp
+++ /home/sojourner/Sources/CAMotics/CAMotics_PATCHED/src/gcode/machine/MoveSink.cpp
@@ -53,14 +53,21 @@
     double feed = rapid ? 10000 : getFeed(); // TODO Get rapid feed from machine
 
     auto &location = getLocation().getStart();
+    unsigned line = location.getLine();
     SmartPointer<string> filename;
 
     if (lastFile.isNull() || *lastFile != location.getFilename())
       lastFile = filename = new string(location.getFilename());
     else filename = lastFile;
 
+    // If no line then pass move count for picking buffer
+    if (line == -1) {
+      line = ++moveCount;
+      filename = new string("GCODE_STREAM");
+    }
+
     Move move(type, start, end, this->time, get(TOOL_NUMBER, NO_UNITS), feed,
-              getSpeed(), location.getLine(), filename, time);
+              getSpeed(), line, filename, time);
 
     this->time += move.getTime();
```

Thanks!

jcoffland, 2023-08-10:
This is a reasonable temporary fix.  I've incorporated something similar.  Thanks!

In general, PRs are preferable.

closed 2023-08-10 by jcoffland

jcoffland, 2023-08-10:
Check out my latest commit.  I figured out how to get filename and line number from TPL and implemented improved picking for TPL.

A remaining issue is that there's really a stack of file/line numbers involved in a particular TPL generated move.  For example, with the ``camotics.tpl`` example that's loaded by default, any clicks in the DXF defined area map somewhere in the built-in library ``dxf.tpl``.  It does not highlight line 13 where the ``dxf.layer_cut()`` call is made.  However, carrying around the entire stack trace for each GCode move could get quite expensive.  CAMotics would also start to look like a Javascript debugger.

Actually, this is also true for GCode.  You can use O-code functions and call to other files in GCode.

A reasonable option might be to trace each move back to the first line in a file that was included in the project.  I probably need to rethink how file locations are handled in CAMotics.  Another thing is that GCode that is loaded by a TPL file yields the correct filename and line but the GCode does not get loaded into the editor. (e.g. ``la_peinture`` example)

50J0uRNeR, 2023-08-11:
That was my original thought, to trace each move back to the original TPL line or at least the generated GCode (and how I might present the GCode in the editor or console).  However it was getting more involved into the internals of CAMotics then I currently understand.

Anyway everything seems to test out good on my system, thanks jcoffland!