CauldronDevelopmentLLC/CAMotics #403: how to add other machines?

tcurdt, 2023-08-22:
I wanted to add a custom machine but run into a few hurdles raising many questions:
- I seems machines can only be added at compile time?
- I suspect the `.tco` file defines the 3D view of the machine - what format is that?
- The positioning of the path/material seems to vary when switching machines. Why is that? How can this be controlled (or may I say fixed)?

[image]
[image]

jcoffland, 2023-08-23:
>    I seems machines can only be added at compile time?

You can add new machines to the ``machines`` directory and restart CAMotics and they should show up.  Currently, you need a ``.json`` and a ``.tco``.

>    I suspect the .tco file defines the 3D view of the machine - what format is that?

It's an arbitrary format that some other developer came up with.   I've considered converting everything to STL files but STL has no provision for storing other info so you'd have to have a separate STL for each part of the machine.  It makes sense to just add the triangles directly to the JSON file rather than referencing an external TCO or STL files.  The TCO contains lists of triangles, lines and vertices.  You'd have to look at the parsing code to get an idea of the format.[^1][^2]

>    The positioning of the path/material seems to vary when switching machines. Why is that? How can this be controlled (or may I say fixed)?

This is messy.  It would be very helpful if we had some sort of machine editor.  You would need to be able to load a triangle mesh (probably from STL) for each axis and the base and then specify home positions (i.e. 3D offsets which put the mesh into the home position), and the relationships between the axes.  On some machines, for example, the x-axis moves with the y-axis and the z-axis moves with both axes.  But that's not always the case.  You could have a stationary z-axis and a moving base or lots of other combinations.  An editor would let you position the parts of the machine at their homes, specify their limits and then save the result as a machine definition.

Once the machine editing is sorted out it would be great to detect collisions or moves beyond the machine's limits.

[^1]:  https://github.com/CauldronDevelopmentLLC/CAMotics/blob/master/src/camotics/machine/MachineModel.cpp#L50
[^2]:  https://github.com/CauldronDevelopmentLLC/CAMotics/blob/master/src/camotics/machine/MachinePart.cpp#L43

jcoffland, 2023-08-23:
I've replace the TCO format with pure JSON.  Now the machine files are in compressed JSON ``.json.bz2`` files.  I completely removed TCO support.

tcurdt, 2023-08-23:
> I've replace the TCO format with pure JSON. Now the machine files are in compressed JSON `.json.bz2` files. I completely removed TCO support.

Wow! That was quick 🙂

> Once the machine editing is sorted out...

But do you really want to add a machine editor? That sounds like a lot of work. Or what exactly do you have in mind there?
Wouldn't it be maybe enough to have a certain number of types of machines that can be parameterised?

I personally wouldn't need a fancy STL type visualisation.
But being able to specify the cutting area is what brought me here. Just my 2 cents.

jcoffland, 2023-08-24:
> But do you really want to add a machine editor? That sounds like a lot of work. Or what exactly do you have in mind there?

I was not thinking about a whole 3D editor but an editor where you could load in the 3D parts created in some other CAD program.

> Wouldn't it be maybe enough to have a certain number of types of machines that can be parameterised?

You do make a good point.  The first thing to do would be to make it possible to just edit the existing machine parameters and save them as your own machine.  Importing the 3D drawings is more involved.

spot1984, 2024-04-18:
It sure would be nice to model machines in CAD packages (e.g. FreeCAD, OpenSCAD, TinkerCAD, Blender, Fusion360, etc.) and export meshes in a common format (e.g. .obj, .step, .stl) that could be converted and assembled into machines either in Camotics or even with something as simple as a python script included in the machines directory that could convert a collection of meshes into a json file that could be edited to finish the machine?  I'm trying to make a suggestion that would balance enabling the community without hand typing vertices, lines, and triangles, and also not distracting Camotics mainstream development

tcurdt, 2024-04-18:
> It sure would be nice to model machines in CAD packages [...] and export meshes in a common format [...] that could be converted and assembled into machines

I looked into the json but TBH I couldn't really make much sense of it yet.

```
{
  "name": "X-Carve",
  "model": "x-carve.tco",
  "tool": [1, 1, 1],
  "workpiece": [0, 0, 0],

  "reverse_winding": true,
  "transform": [
    [1, 0, 0, 0],
    [0, -1, 0, 75],
    [0, 0, 1, 0],
    [0, 0, 0, 1]
  ],

  "parts": {
    "x": {
      "init": [0, 0, 0],
      "home": [0, 0, 0],
      "min": [0, 0, 0],
      "max": [9, 0, 0],
      "movement": [1, 1, 0]
    },

    "y": {
      "init": [0, 0, 0],
      "home": [0, 0, 0],
      "min": [0, 0, 0],
      "max": [0, 0, 0],
      "movement": [0, 1, 0]
    },

    "z": {
      "init": [0, 0, 0],
      "home": [0, 0, 0],
      "min": [0, 0, 0],
      "max": [0, 0, 0],
      "movement": [1, 1, 1]
    }
  }
}
```

Why doesn't this have dimensions but is mostly 1 based?
Why is transform a 4x4 matrix?

I would love to add my customized PrintNC somehow. It would be super cool if you could give some guidance, @jcoffland

jcoffland, 2024-04-19:
https://www.brainvoyager.com/bv/doc/UsersGuide/CoordsAndTransforms/SpatialTransformationMatrices.html

tcurdt, 2024-04-19:
>> Why is transform a 4x4 matrix?
> https://www.brainvoyager.com/bv/doc/UsersGuide/CoordsAndTransforms/SpatialTransformationMatrices.html

I see - rotate and translate.

But how are the dimensions defined? Through the related 3d model I assume?

jcoffland, 2024-04-20:
It's been a long time since I looked at this.  The source code is the best place to find out how this all works.

tcurdt, 2024-04-20:
I can't find the `x-carve.tco` referenced in the json.

I've checked https://github.com/CauldronDevelopmentLLC/CAMotics/tree/master/src/camotics/machine
but I don't see where the 3d model of the machine comes in yet.

tcurdt, 2024-04-26:
> CAMotics no longer uses the TCO format. The machine files are bzip2 compressed JSON. See the `machines` directory.

Sure - but neither the old TCO format nor the JSON format has the 3D Model information.

jcoffland, 2024-04-26:
The JSON files contain all the part data.  The parts contain ``lines`` and ``mesh`` which describe the shape.

tcurdt, 2024-04-26:
This is the `shapeoko_3.json`. I don't see any lines or mesh. 🤔

```
{
  "name": "Shapeoko 3",
  "model": "shapeoko_3.tco",
  "tool": [1, 1, 1],
  "workpiece": [0, 0, 0],

  "reverse_winding": true,
  "transform": [
    [1, 0, 0, 0],
    [0, -1, 0, 75],
    [0, 0, 1, 0],
    [0, 0, 0, 1]
  ],

  "parts": {
    "x": {
      "init": [0, 0, 0],
      "home": [0, 0, 0],
      "min": [0, 0, 0],
      "max": [9, 0, 0],
      "movement": [1, 1, 0]
    },

    "y": {
      "init": [0, 0, 0],
      "home": [0, 0, 0],
      "min": [0, 0, 0],
      "max": [0, 0, 0],
      "movement": [0, 1, 0]
    },

    "z": {
      "init": [0, 0, 0],
      "home": [0, 0, 0],
      "min": [0, 0, 0],
      "max": [0, 0, 0],
      "movement": [1, 1, 1]
    }
  }
}
```

jcoffland, 2024-04-26:
That's the old file.

https://github.com/CauldronDevelopmentLLC/CAMotics/blob/master/machines/shapeoko_3.json.bz2

tcurdt, 2024-04-26:
Just realized the I am running an old version (1.2.0). I guess that's why.

spot1984, 2024-11-18:
1.2.0 is the latest release, so still using the .tco format until the next release.

spot1984, 2024-12-08:
@tcurdt, @jcoffland I wrote a python script to create Camotics machines from X3D .xhtml files for my local Workshop 88 Makerspace.  I thought you might like it. 

Here is the script https://github.com/Workshop88/camotics_machine_maker
I tried to add as much documentation to the code and the github repo, including instructions how to configure the .json and how to copy the files to where Camotics can use them.

I hope this works for you, at least until the next Camotics release.  

**_Is there somewhere more public I should post this?_**

jcoffland, 2024-12-09:
Wow that is awesome work.  Thank you for sharing. You could post a message on the CAMotics users group. https://groups.google.com/g/camotics-users