CauldronDevelopmentLLC/CAMotics #404: Newer Version Build for usage with FreeCAD

ModuloFS, 2023-08-27:
Hello,

would it be possible to release an windows build of the current source code to use it directly out of FreeCAD? 
Or is an estimation available until a new version of camotics would be released?

jcoffland, 2023-08-27:
I've been wanting to release a new build but I've not been able to find time to test the latest build in Windows and macOS.  I'm thinking of doing an alpha/beta release and asking others to test it outside of Linux for me.

ModuloFS, 2023-08-28:
If its helpful, i can also test the beta release on Windows.

bongiben, 2023-08-28:
would like to try on a Mac

jcoffland, 2023-08-29:
Ok, I'll try to find some time in the next few weeks to make a new build.  I need to get my build machines spun back up, work through any build problems on macOS and Windows then I can make an alpha release.  I'll update you when it's ready.

lkarlslund, 2023-09-12:
Consider using the Github actions for doing automated builds :-)

luzpaz, 2023-10-20:
@jcoffland Is there an ETA on a release date ?

JohnOCFII, 2023-10-20:
Would also be interested in a macOS arm native version too, as FreeCAD supports that native M1/arm environment.

ModuloFS, 2023-12-15:
Hello, any news on this topic? :)

jcoffland, 2023-12-16:
No ETA.  I haven't been able to find the time for this.  I've been too busy with travel and paid work.  It's a couple of days work which I will fit in as soon as I can.

RiccaDS, 2024-03-20:
Interested in this too, user of FreeCAD in Linux here

tcurdt, 2024-04-26:
Also happy to test on macOS.

luzpaz, 2024-10-25:
Any progress ?

tcurdt, 2025-07-07:
What makes it such a big effort - anything we could help with?

jcoffland, 2025-07-07:
Building and testing on Windows, Linux and macOS takes a lot of time.  I work in Linux so Linux is not a big problem.  I've setup a VM for Windows builds but there are always issues, including just maintaining the OS.  Then I have an account with macstadium where I build the macOS version but there are often connection problems and again the building and debugging takes time.  I'm actually using a version of buildbot to that makes this all a bit easier but it's never automatic.

Also, to release packages that actually install on Windows and macOS, without making the user jump through a lot of hoops, you need to sign the packages.  I've got code signing keys for both Windows and macOS.  The macOS signing process is particularly difficult but I do have it working.  I'm not willing to sign packages that someone else builds for security reasons.

Ideally, the Linux version would just be integrated in to the Debian package system but someone would have to step up as the package maintainer.

If there's a better way I'm open to it.

I've started the process several times over the last few years and not finished it.  It takes several days of work to get it done and I've not been able to justify that time when I've got a lot of other paid work to do.

Finally, this is Open-Source so there's no reason why someone else cannot build and release this software.

tcurdt, 2025-07-07:
That sounds all very manual.
Have you ever tried setting it up so the CI to does the builds?
Maybe we could contribute that and lower the burden for you that way?

jcoffland, 2025-07-07:
It is basically the same as a CI system but the builds are not triggered automatically by the source control.  CI systems take maintenance.  CI systems on three OS take more maintenance.  There's a lot of manual work to make it work automatically.  Contributions are always welcome.