CauldronDevelopmentLLC/CAMotics #405: Debian 12 libssl and v8_compress_pointers (Embedder-vs-V8 build configuration mismatch)

sbeaugrand, 2023-09-04:
To fix libssl and v8_compress_pointers on debian 12 bookworm, i have to do this :

# Build with docker
```
docker build -t camotics .
docker run -v $PWD:/pwd camotics cp CAMotics/camotics_1.3.0_amd64.deb /pwd/

sudo apt install ./camotics_1.3.0_amd64.deb
docker container prune
docker rmi camotics
docker rmi debian:stable-slim
```

# Build without docker
```
sudo apt install equivs
equivs-build camotics-control
sudo apt install ./camotics-build-deps_1.0_all.deb

git clone https://github.com/CauldronDevelopmentLLC/cbang
git clone https://github.com/CauldronDevelopmentLLC/CAMotics
scons -C cbang v8_compress_pointers=false
export CBANG_HOME=$PWD/cbang
cd CAMotics
sed -i 's/libssl1.1/libssl3/' SConstruct
scons
scons package

sudo apt install ./camotics_1.3.0_amd64.deb
sudo apt remove camotics-build-deps
sudo apt autoremove
```

# Dockerfile
```
FROM debian:stable-slim

RUN apt-get update\
 && apt-get -y install\
 scons\
 build-essential\
 libqt5websockets5-dev\
 libqt5opengl5-dev\
 libnode-dev\
 libglu1-mesa-dev\
 pkgconf\
 git\
 python3-six\
 python3-setuptools\
 qttools5-dev-tools\
 && git clone https://github.com/CauldronDevelopmentLLC/cbang\
 && git clone https://github.com/CauldronDevelopmentLLC/CAMotics\
 && scons -C cbang v8_compress_pointers=false\
 && export CBANG_HOME=$PWD/cbang\
 && cd CAMotics\
 && sed -i 's/libssl1.1/libssl3/' SConstruct\
 && scons\
 && scons package
```

# camotics-control file
```
Package: camotics-build-deps
Depends: scons, build-essential, libqt5websockets5-dev, libqt5opengl5-dev, libnode-dev, libglu1-mesa-dev, pkgconf, git, qttools5-dev-tools
```

sakesfar, 2024-02-05:
Hi! 

I am a newbie to Linux. I have Debian 12 and need to build Camotics! However, I could not build it as written in readME. I encountered multiple issues. Last errors were identical to the discussed here . 

So, to clarify, do I need to go through all the steps mentioned here ? 
So far, when typing  `docker build -t camotics .` I have :
`RROR: "docker buildx build" requires exactly 1 argument.
See 'docker buildx build --help'.

Usage:  docker buildx build [OPTIONS] PATH | URL | -

Start a build`

Also, since I might get issues ahead. I want to be sure to delete and install back (build).
I tried to use `make clean` but the terminal also gave erros citing some Scons/somefile.py related mistakes at different lines. It blew my mind :) 

How to delete then ? 

I am sorry for these questions here. Just a new person in Linux getting hammered by these issues :)

atorrey, 2024-03-09:
I tried using the above directions on my Debian Trixie machine, and was successful - but had some issues, below are my notes on what I encountered - my notes begin w/ ":"
    :STARTED in ~/CAMotics

sudo apt install equivs
equivs-build camotics-control
    :Would not run - no such file error - did "touch camotics-control" in ~/CAMotics, it then worked, but w/ message: : 
    :Attention, the package has been created in the current directory, not in ".." as indicated by the message above!

sudo apt install ./camotics-build-deps_1.0_all.deb
    :got error: E: Unsupported file ./camotics-build-deps_1.0_all.deb given on commandline

git clone https://github.com/CauldronDevelopmentLLC/cbang
git clone https://github.com/CauldronDevelopmentLLC/CAMotics
    :SKIPPED - as did a few hours ago...
    
    :Switched to ~/cbang
scons -C cbang v8_compress_pointers=false
    :ran w/o errors
    
    :Note, must do this in ~ - if done in /CAMotics or /cbang, CAMotics compile will fail to find cbang
export CBANG_HOME=$PWD/cbang
    :should be ~/cbang (or whatever the cbang directory is)
cd CAMotics
sed -i 's/libssl1.1/libssl3/' SConstruct
scons
scons package

This resulted in a successful install that opened from my KDE menu.

Lubensius, 2024-10-13:
Thx. Compiled it under Ubuntu 24.04.1LTS with change to libssl3 and disabled V8 and it seems to work,
I skipped
```
sudo apt install equivs
equivs-build camotics-control
sudo apt install ./camotics-build-deps_1.0_all.deb
```
as I don't think it is needed here.

seisdr, 2024-12-13:
It doesn't compile no more!