CauldronDevelopmentLLC/CAMotics #407: Build broken due to missing boost header

arunabha, 2023-10-08:
The build for the master branch seems to be broken.

I followed instructions on the Camotics Github page(built C! etc). However when I try to build Camotics, I'm getting the following error

```
g++ -o build/camotics/sim/Workpiece.o -c -faligned-new -std=c++14 -fsigned-char -Wno-deprecated-declarations -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DHAVE_CBANG -DUSING_CBANG -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_NETWORK_LIB -DQT_WIDGETS_LIB -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -DCBANG_WRAP_GLIBC -I/home/arun/dev/cbang/src -I/home/arun/dev/cbang/include -I/home/arun/dev/cbang/src/boost -I/usr/include/v8 -Isrc -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtOpenGL -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtWebSockets -Ibuild -Isrc src/camotics/sim/Workpiece.cpp
src/camotics/sim/ToolPathTask.cpp:56:10: fatal error: boost/iostreams/device/file.hpp: No such file or directory
   56 | #include <boost/iostreams/device/file.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
scons: *** [build/camotics/sim/ToolPathTask.o] Error 1
scons: building terminated because of errors.
```

Looking into it a bit further, it looks like the file boost/iostreams/device/file.hpp was removed in this [recent commit] (https://github.com/CauldronDevelopmentLLC/cbang/commit/74334acb197533a19e3356aa5623cfcd7875b2b8#diff-f2fe6d18f5f3a76ef83dbbc0c18cf48c8d6ddf7bfed47f51a9b9b5d66f322f50)

Any suggestions on how to fix this?

jcoffland, 2023-10-08:
I've committed a fix.

luzpaz, 2023-10-20:
@arunabha can you reproduce ?

arunabha, 2023-10-20:
Confirmed. I'm able to build the master branch.

closed 2023-10-20 by jcoffland