CauldronDevelopmentLLC/CAMotics #409: V8 error on Debian stable

leventelist, 2023-11-01:
I built the code on Debian stable, and I get this when I try to run it:

```
#
# Fatal error in , line 0
# Embedder-vs-V8 build configuration mismatch. On embedder side pointer compression is ENABLED while on V8 side it's DISABLED.
#
#
#
#FailureMessage Object: 0x7ffd7680fc70
==== C stack trace ===============================

    /lib/x86_64-linux-gnu/libnode.so.108(v8::base::debug::StackTrace::StackTrace()+0x16) [0x7fc88ce9aa66]
    /lib/x86_64-linux-gnu/libnode.so.108(+0xd506dc) [0x7fc88c5506dc]
    /lib/x86_64-linux-gnu/libnode.so.108(V8_Fatal(char const*, ...)+0x167) [0x7fc88ce8b8d7]
    /lib/x86_64-linux-gnu/libnode.so.108(v8::V8::Initialize(int)+0x43) [0x7fc88c576ee3]
    ./camotics() [0x497894]
    /lib/x86_64-linux-gnu/libc.so.6(+0x271ca) [0x7fc8898461ca]
    /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7fc889846285]
    ./camotics() [0x49e8e1]
Aborted
```

jcoffland, 2023-11-01:
C! builds with v8 pointer compression disabled by default.  You can turn it on by building C! with ``scons v8_compress_pointers=1``.  However, the error above indicates that C! was built with pointer compression on but the library has it turned off, unless I'm misunderstanding it.  It's possible that your building with one set of v8 headers and linking with a non-matching v8 library.  I.e. the v8 headers might turn pointer compression on despite the library being built with out it.

jcoffland, 2023-11-01:
See #340

sbeaugrand, 2023-11-21:
See #405