CauldronDevelopmentLLC/CAMotics #410: More of a question, can Camotics support multiple cut bodies?

ShamanTcler, 2023-11-16:
I am thinking of brute forcing a problem here. The problem: It is not unusual for a tool to run into a clamp. 

If Camotics would show where it milled through a clamp, people might save a bit and clamp.

Even if it looped over each body separately this could be useful.

jcoffland, 2023-11-16:
This has been discussed a few times.  See #358 for example.  Also mentioned in #403.  It's not too difficult depending on how the shape of the fixtures is defined.  The simplest thing to do would be to treat all fixtures as blocks and assume you never want to cut inside the block.  Detecting collisions accurately with more complex shapes becomes more difficult.

ShamanTcler, 2023-12-06:
Once this was made am "enhancement" I did not reply, but should have. Simple blocks are just fine.  Thanks for the consideration.

Kerpom, 2024-08-26:
On many machines, the rapid path is not interpolated. This means all axis go with the max speed independantly from each others.
Let's consider the following example:
G0 X0 Y0 Z100.0
G0 X100.0 Y40.0
In this case, the path will look in two steps.
First: both axis at max speed, the result is X40 Y40
Then the X axis keeps going up to X100 Y40.
It should be considered in case of collision detection feature.