CauldronDevelopmentLLC/CAMotics #412: Spindle warm up GCode ... yup no axis move

ShamanTcler, 2024-01-09:
This may be a "corner case" but I have written a spindle warm up that does not move any of the axis. I expected the code to run, and watch the RPMs to go up and down with time.... but I get nothing.

**Code snippet:**
```
(Safe block)
(G20 Units: inches)
(G90 Absolute distance mode)
(G17 work on in the X & Y plane)
(G80 cancel all active canned cycles)
(G40 turn off cutter compensation)
(G49 turn off tool length compensation)
G20 G90 G17 G80 G40 G49 

(Work offset: G54)
G54

(4500 RPM, turn on spindle CW)
S4500 M03 
(Dwell 60 seconds)
G04 X60
(Stop spindle)
M05

(end the program)
M30
```

jcoffland, 2024-01-09:
CAMotics does not currently simulate dwells.  Also, the correct code for a dwell is ``G04 P60``.

ShamanTcler, 2024-01-09:
Oh how I wish there was an ANSI standard for GCode. Seem some dialects of GCode support something like 

```
G04 X3.5 (dwell 3.5 seconds)
```

as far as you are concerned, do you support:
```
G04 P3.5 (dwell 3.5 seconds)
```

or must it be:
```
G04 P3500 (dwell 3.5 seconds)
```

jcoffland, 2024-01-09:
Only the middle one.

The GCode supported by CAMotics is described here: https://camotics.org/gcode.html  There is a GCode standard, RS274/NGC.  Not everyone follows it.