CauldronDevelopmentLLC/CAMotics #416: Compile fail on Debian Trixie

atorrey, 2024-03-08:
I have attempted multiple times to follow the directions on the cbang and  CAMotics pages to compile the source, and keep getting compile errors.  I'm not a programmer, so am not sure about what the messages I'm seeing mean.  Cbang appears to compile without errors, and the test harness passes all tests per it's report.

As far as I can tell, I've done all the steps in the readme files, and the system is completely updated 

This is the output when I attempt to compile.  If you can suggest what I need to fix (or fix the bug if it's not me....) Please let me know.  If you need more info, please let me know.

`root@fancybox:/home/atorrey/CAMotics# scons strict=0
scons: Reading SConscript files ...

scons: warning: QT5DIR variable is not defined, using moc executable as a hint (QT5DIR=/usr)
File "/home/atorrey/CAMotics/config/qt5/__init__.py", line 449, in _detect
Compiler: gcc (gnu)
Platform: posix
Mode: gnu
Arch: x86
GCC Version: 13
Jobs: 12
Checking for C library pthread... yes
Checking for C header file zlib.h... yes
Checking for C library z... yes
Checking for C header file bzlib.h... yes
Checking for C library bz2... yes
Checking for C header file expat.h... yes
Checking for C library expat... yes
Checking for C header file pthread.h... yes
Checking for C library pthread... yes
Checking for C library dl... yes
Checking for C library sqlite3... yes
Checking for C header file sqlite3.h... yes
Checking for C function sqlite3_backup_init()... yes
Checking for C header file yaml.h... yes
Checking for C library yaml... yes
Checking for C library snappy... yes
Checking for C library leveldb... yes
Checking for C++ header file leveldb/db.h... yes
Checking for C header file mysql/mysql.h... yes
Checking for C library mariadbclient... yes
Checking for C function mysql_real_connect_start()... yes
Checking for C library lz4... yes
Checking for C function clock_gettime()... yes
Checking for C header file event.h... yes
Checking for C library event... yes
Checking for C++ header file re2/re2.h... yes
Checking for C library re2... yes
Checking for C function epoll_create1()... yes
Checking for C library dl... yes
Checking for C header file openssl/ssl.h... yes
Checking for C library crypto... yes
Checking for C library ssl... yes
Checking for openssl version >= 1.1.0...yes
Checking for C++ header file v8.h... yes
Checking for C++ header file libplatform/libplatform.h... yes
Checking for C library v8_monolith... no
Checking for C library v8... yes
Checking for C library v8_libplatform... yes
Checking for C header file pthread.h... yes
Checking for C library pthread... yes
Checking for C header file systemd/sd-bus.h... no
Checking for C header file valgrind/valgrind.h... no
Checking for C library cbang-boost... yes
Checking for C library cbang... yes
Checking for C++ header file cbang/Exception.h... yes
Checking for C header file Python.h... yes
Checking for C function Py_Initialize()... yes
Checking for C library GL... yes
Checking for C header file cairo/cairo.h... yes
Checking for C library cairo... no
Checking for C++ header file dxflib/dl_dxf.h... yes
Checking for C library dxflib... no
Need library dxflib(cached) error: no result
Checking for C library stdc++... yes
scons: done reading SConscript files.
scons: Building targets ...
g++ -o build/camotics.o -c -faligned-new -std=c++14 -fsigned-char -Wno-deprecated-declarations -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DHAVE_CBANG -DUSING_CBANG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -DCBANG_WRAP_GLIBC -I/home/atorrey/cbang/src -I/home/atorrey/cbang/include -I/home/atorrey/cbang/src/boost -I/usr/include/v8 -Isrc -I/usr/include/python3.11 -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtOpenGL -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtWebSockets -Ibuild -Isrc src/camotics.cpp
/usr/bin/rcc -name camotics qt/camotics.qrc -o build/qrc_camotics.cpp
/usr/bin/lrelease -qm build/i18n/camotics_de.qm languages/camotics_de.ts
lrelease: could not exec '/usr/lib/qt5/bin/lrelease': No such file or directory
build_function(["build/build_info.cpp"], [])
g++ -o build/camotics/Application.o -c -faligned-new -std=c++14 -fsigned-char -Wno-deprecated-declarations -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DHAVE_CBANG -DUSING_CBANG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -DCBANG_WRAP_GLIBC -I/home/atorrey/cbang/src -I/home/atorrey/cbang/include -I/home/atorrey/cbang/src/boost -I/usr/include/v8 -Isrc -I/usr/include/python3.11 -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtOpenGL -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtWebSockets -Ibuild -Isrc src/camotics/Application.cpp
g++ -o build/camotics/CommandLineApp.o -c -faligned-new -std=c++14 -fsigned-char -Wno-deprecated-declarations -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DHAVE_CBANG -DUSING_CBANG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -DCBANG_WRAP_GLIBC -I/home/atorrey/cbang/src -I/home/atorrey/cbang/include -I/home/atorrey/cbang/src/boost -I/usr/include/v8 -Isrc -I/usr/include/python3.11 -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtOpenGL -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtWebSockets -Ibuild -Isrc src/camotics/CommandLineApp.cpp
g++ -o build/camotics/ConcurrentTaskManager.o -c -faligned-new -std=c++14 -fsigned-char -Wno-deprecated-declarations -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DHAVE_CBANG -DUSING_CBANG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -DCBANG_WRAP_GLIBC -I/home/atorrey/cbang/src -I/home/atorrey/cbang/include -I/home/atorrey/cbang/src/boost -I/usr/include/v8 -Isrc -I/usr/include/python3.11 -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtOpenGL -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtWebSockets -Ibuild -Isrc src/camotics/ConcurrentTaskManager.cpp
g++ -o build/camotics/Grid.o -c -faligned-new -std=c++14 -fsigned-char -Wno-deprecated-declarations -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DHAVE_CBANG -DUSING_CBANG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -DCBANG_WRAP_GLIBC -I/home/atorrey/cbang/src -I/home/atorrey/cbang/include -I/home/atorrey/cbang/src/boost -I/usr/include/v8 -Isrc -I/usr/include/python3.11 -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtOpenGL -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtWebSockets -Ibuild -Isrc src/camotics/Grid.cpp
g++ -o build/camotics/SHA256.o -c -faligned-new -std=c++14 -fsigned-char -Wno-deprecated-declarations -ffunction-sections -fdata-sections -O3 -funroll-loops -fno-pie -fPIC -DNDEBUG -D_REENTRANT -DHAVE_CBANG -DUSING_CBANG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -DCBANG_WRAP_GLIBC -I/home/atorrey/cb [... cut ...] oost -I/usr/include/v8 -Isrc -I/usr/include/python3.11 -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtOpenGL -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtWebSockets -Ibuild -Isrc src/camotics/contour/CorrectedMC33.cpp
scons: building terminated because of errors.
root@fancybox:/home/atorrey/CAMotics#
`

jcoffland, 2024-03-08:
This is the problem:

```
lrelease: could not exec '/usr/lib/qt5/bin/lrelease': No such file or directory
```
``lrelease`` is part of ``qttools5-dev-tools``.

closed 2024-03-08 by jcoffland

atorrey, 2024-03-09:
Thank you, that seems like at least part of the problem as the compile seems to have run without errors.  However I'm still having problems getting the package to install.  Will look to see if the issues are already existing, or open a new one for the specific problem I'm getting.