CauldronDevelopmentLLC/CAMotics #422: How to compile for non-debian based system?

RWL-69, 2024-07-03:
I would like to be able to use CAMotics again but there are no RPM's available for current system.  Have tried to install the rpm's as directed from pkgs.org, that I have found, but keep having dependency issues due to requirements for older versions.  This may be why the maintainers dropped the package.

I downloaded the source and it only describes how to make a debian package.  I would be happy just to have a binary file I can load.

I am not a programmer but have done some compiling in the past.

jcoffland, 2024-07-03:
The build instructions are essentially the same as in the README.md for Debian systems but you're on your own figuring out which packages on your distro provide the same dependencies.

RWL-69, 2024-07-03:
DEB packages are what teh instructions say from what I am reading. 

> 
> ## Building CAMotics
> Clone the CAMotics git repository and build the software using scons:
> 
>     git clone https://github.com/CauldronDevelopmentLLC/CAMotics.git
>     cd CAMotics
>     scons
> 

I guess the question is what does this command leave you with? 

I will try to build it in the next few days.

jcoffland, 2024-07-03:
https://github.com/CauldronDevelopmentLLC/CAMotics?tab=readme-ov-file#using-camotics

RWL-69, 2024-07-04:
I read through some of the reason that CAMotics is not on Fedora is due to C! and packaging.  It was dropped since Cbang was an issue and strange dependencies for stuff that wasn't needed.  I would have to spend more time reading through the bug report.  I hope to try to compile on the weekend.  Starting with C!.

It would be nice to see a flatpack or appimage of CAMotics.

For reference.

> [Bug 2129303](https://bugzilla.redhat.com/show_bug.cgi?id=2129303) - Review Request: cbang - The C! or cbang library is a collection of C++ utility libraries