CauldronDevelopmentLLC/CAMotics #428: `print`s missing from gcode export

Erhannis, 2024-09-12:
I was using `print` to insert gcode to work around some vagaries of my CNC, and it seems like I get different output when I simulate and export gcode vs when I run tplang directly.  TPL:
```
print("G90\n");
rapid({x: 10});
rapid({x: 20});
```
Simulate then export gcode from camotics:
```
G21
(File: /home/erhannis/3dPrinting/created/laser_vectors_folder/benchmark/scratch.tpl)
N2 G0 X10.
N3 G0 X20.
M2
```
tplang:
```
G21
G90
(File: scratch.tpl)
N2 G0 X10.
N3 G0 X20.
M2
```

I can work around it, of course, just by using tplang directly, but I would have expected the outputs to be the same.  Is the difference intentional?

Btw, in general, is there an "official" way of inserting raw gcode?  `gexec` appears to process the gcode and discard things that don't affect the simulator, which then bites me when my cnc doesn't behave the same as the simulator (e.g. wanting particular flags set via gcode, or apparently demanding a dedicated "G1 S1000" to turn on the cutter for the first time, etc.)  I've settled on using `print`, so far, but if the commands WERE to have an effect on the simulator, it'd be nice to not have to worry about ensuring the commands end up in both the sim and the real machine and/or duplicate across both gexec and print, etc.  Like, if there were a command that parsed gcode, but also made sure to output that same gcode into the output.  I don't know if that's compatible with the underlying system or not.

closed 2024-09-16 by jcoffland

Erhannis, 2024-09-17:
I did start with `gexec` - unfortunately, as I mentioned, 
> gexec appears to process the gcode and discard things that don't affect the simulator, which then bites me when my cnc doesn't behave the same as the simulator
For instance, despite
> G90 is the default anyway
my CNC interpreted the gcode in G91 mode, and needed G90 set explicitly.  Unfortunately, `gexec` was unable to do this, with `gexec("G90")`; producing no such command in the output.  Hence my use of `print`.

Given that sometimes weird workarounds are the most practical solution to a problem, it seems wise to provide a mechanism to enable them, even if they're discouraged.  Perhaps a `gprint` function?  I think I can add the code myself, if you'd like.

Also - the documentation states "Note that print() writes to the same output stream as the gcode functions so if you don't want to interfere with the gcode output you should wrap text in parentheses () so that it is treated as a gcode comment."  So if instead, as you say, "It writes to a different output stream", it seems that either this behavior should be changed, or the documentation should be updated.

jcoffland, 2024-09-17:
I must be wrong about ``print()`` using different output stream.  The reason ``gexec("G90")`` does not do anything is because in CAMotics ``G90`` is the default and CAMotics always simplifies it's output.  It never outputs codes like ``G90`` or ``G91``.  If you use incremental mode it will alter the following GCode commands but CAMotics will convert them to absolute moves in the output.

I tried to find a work around but failed.  I think the best solution would be to have CAMotics output ``G90`` in its initialization.  Currently it only outputs ``G21`` but it should probably include all of the state assumptions it is making.  Actually, there is a note already here: https://github.com/CauldronDevelopmentLLC/CAMotics/blob/master/src/gcode/machine/GCodeMachine.cpp#L91.  One danger though is that some machines may interpret some codes differently.  What we really need is proper machine definitions in CAMotics but this is a lot more work.

jcoffland, 2024-09-17:
I've committed a change that adds ``G90`` to the output.

Erhannis, 2024-09-18:
While I agree that having CAMotics explicitly state its default assumptions is a good idea, I don't think it solves the underlying problem that sometimes users have eccentric machines that want specific things that the CAMotics processing code can't reasonably support, explicitly.  In my case, G90 was just one of the things - another was that for some reason my machine (a laser cutter) only performed a stationary laser turn-on if I did `G1 S1000`, as opposed to `M3 S1000`.  (`M3 S1000` and then moving would turn on the laser, but I wanted to perform an initial pierce.)  `S` isn't one of the parameters CAMotics supports for cut commands, nor do I really think it should be.  (Barring some kind of general "tack on whatever additional parameters you need" argument, and I'm not convinced that's good, either.)  I still think the most flexible solution is to permit the user to insert arbitrary text into the output stream when nothing else suffices - perhaps this `gprint` function I've added would do the trick?  https://github.com/Erhannis/CAMotics/tree/feature/gprint  It's basically just the `comment` function, without the comment markers.

Btw, after looking at the code, I think `print` is odd because it's not a special function like `comment` or `message`; I suspect it's just a built-in function.  Its output, when you export gcode from CAMotics, ends up in the console logs rather than the exported file - it just so happens that if you use tplang, the generated file is itself output to the console logs, so they end up in the same place.

jcoffland, 2024-09-18:
Your ``gprint()`` approach does work to inject some code in to the output GCode from TPL.  However, it would be lost if you passed the code through any further CAMotics GCode processor and CAMotics simulation will ignore it.

There is a facility to add special procedures which override specific GCodes.  We use this facility to program tool change procedures for the Buildbotics CNC controller.  Currently it's only supported via the path planner but this could be used to implement the behavior of specific machines.  ``GCodeRunner`` reads ``overrides`` from the config.  There's no reason why this overrides could not be supported elsewhere.  However, this is part of generalized machine modeling which is not yet supported properly in CAMotics.

Erhannis, 2024-09-21:
Hmm, I see.  Well, I guess in the meantime I'll keep using `print` and so forth; thanks.