CauldronDevelopmentLLC/CAMotics #438: camotics: error while loading shared libraries: libv8.so.3.14.5

reset12, 2024-12-22:
Had the same error camotics: error while loading shared libraries: libv8.so.3.14.5 on Ubuntu 24.04 and 24.10.
It seems an old error that is more than once reported. Is there a chance to repair that ?

hesperaux, 2025-06-22:
See my comment on https://github.com/CauldronDevelopmentLLC/CAMotics/issues/215

It's a hack but it will get it to run. It looks like the developers are not going to fix this issue.

SirGeekALot, 2025-07-10:
Thanks for the reply.  I tried again, doing the following:

`sudo apt remove camotics`
(successfully removed)

`sudo apt install libnode-dev`

Response:
`libnode-dev is already the newest version (20.18.1+dfsg-1ubuntu2).`

`sudo dpkg -i camotics_1.2.0_amd64.deb`
(successfully installed)

`cd /usr/bin`
`./camotics`

`camotics: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory`

hesperaux, 2025-07-11:
> Thanks for the reply.  I tried again, doing the following:
> 
> `sudo apt remove camotics`
> (successfully removed)
> 
> `sudo apt install libnode-dev`
> 
> Response:
> `libnode-dev is already the newest version (20.18.1+dfsg-1ubuntu2).`
> 
> `sudo dpkg -i camotics_1.2.0_amd64.deb`
> (successfully installed)
> 
> `cd /usr/bin`
> `./camotics`
> 
> `camotics: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory`
> 
> EDIT: wait. forgot to link...
> 
> `sudo ln -s libv8.so libv8.so.3.14.5`
> 
> `./camotics`
> 
> same error:
> `camotics: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory`

Oh it's just missing openssl. You probably have openssl3 installed, but check for openssl1 and install that.
The package is probably called openssl-1 or libopenssl-1 or similar (apt search openssl).

If you already have it installed, you might have another version of openssl 1.x. In that case you can use the same kind of hack to point libssl.so.1.1 to libopenssl.so.1.

bombela, 2025-07-17:
For what platform this deb package is built for? I symlinked libv8, libssl, and libcrypto. 

[image]

SirGeekALot, 2025-07-17:
```
cd /usr/bin
ls -l openssl
-rwxr-xr-x 1 root root 1132096 Apr  3 01:48 openssl

openssl -v
OpenSSL 3.4.1 11 Feb 2025 (Library: OpenSSL 3.4.1 11 Feb 2025)

```
I definitely have openssl installed. I just don't know what ln (link) command I should use. Help?

hesperaux, 2025-07-18:
Most likely it's built for debian (that's what I'm using).
Unfortunately if your package manager doesn't have openssl1 you will have to build it yourself or risk it with a .deb from the internet. You could try taking the deb from debian 12 but it may cause conflicts.

The best solution would be to compile openssl v1 from source, then manually install a symlink. Sounds like a good question for chatgpt at this point! 😅

You can also build the so file and put it into the same directory as the executable. This usually works too (instead of making a symlink in system directories).

hesperaux, 2025-07-18:
> I am on debian 12.11 and I only have libsslv3 available. I don't think
> opensslv1 is easily available, probably because it's most often used
> over the network and it isn't safe. But I could be mistaken.

Well now I am wondering why it works in my machine... 🤔
And you're right about the rest of course. If it wasn't being used in a networkless cam sim I would be skeptical. 
Actually, what IS openssl used for in this software?